tfabris/CrowCam #64: Obtain OAuth credentials

deltasierra, 2020-05-25:
There doesn't seem to be an `Application type: Other` when creating OAuth credentials.  

The options are: 
- Web Application
- Android
- Chrome 
- iOS
- TV and Limited Input devices 
- Desktop app
- Universal Window Platform 

I tried Web Application but got an error.  Any suggestions?

tfabris, 2020-05-25:
Indeed. Blast, they changed the menu. I'm poking at it to figure out which one is the correct one now. Give the "Desktop" one a try and see if that works?

tfabris, 2020-05-25:
I went through the setup steps again on an experimental account, choosing "Desktop app" at that Google screen, and it seemed to work for me. Give that a try and let me know if it worked? If it worked, you can close this bug by using the appropriate button (if GitHub lets you, not sure if it will or not).

Thanks very much for bringing this to my attention.

deltasierra, 2020-05-27:
That seemed to work, thank you so much for the work on putting this together!

closed 2020-05-27 by deltasierra

tfabris, 2020-05-27:
@deltasierra - Cool. I've updated the code with an extra feature today, so feel free to pick up that version as well. It basically does a quick stream bounce at the top of every hour, to try to fix issue #14. 

The new version has a new setting in "crowcam-config", so if you want that version then you'll need to overwrite that config file with the new version, so make sure to save off any changes you made to that file.  (Note to self: Fix it so that there can be a true user configuration file that isn't overwritten...)

Let me know if these scripts are useful to you. I don't think anyone else besides you and me are using this code. :)

deltasierra, 2020-05-27:
Awesome, I'll pull that down tomorrow.  

Another question for you.  Is there an easy way to disable sunrise/sunset in order to run 24x7?  I was thinking I could comment out lines 775-781 of `CrowCam.sh` but I'm haven't looked deeply on where else that would impact.

tfabris, 2020-05-27:
@deltasierra - I think I'll try to implement sunrise/sunset control as a feature in the configuration file shortly.

tfabris, 2020-05-27:
@deltasierra - New issue opened for this feature: #65 - Move discussion there if you are interested.