tfabris/CrowCam #96: CrowCam Controller - Problem obtaining sunrise/sunset from Google

quog, 2025-01-29:
Hi Tony,

I have been getting this message of late running my copy of CrowCam (DSM7):

CrowCam Controller - Problem obtaining sunrise/sunset from Google. Falling back to previously saved values.

Rather than try to debug the Google output parsing, I have made a small change to get the sunrise and sunset times from api.sunrisesunset.io - which is a free JSON style API.

I've modified two files - crowcam-config - where I added latitude and longitude values (from Google Maps.)

```
$ diff -u crowcam-config\ -\ Backup crowcam-config
--- "crowcam-config - Backup"   2024-07-15 13:13:58.168000000 +0800
+++ crowcam-config      2025-01-29 15:14:28.515000000 +0800
@@ -180,7 +180,9 @@
 # Set this value to either blank ("") or your location. No spaces.
 #
 location="Perth,WA"
-
+lat=-31.998903949338903
+lng=116.00531711164594
+

 # Difference, in minutes, that you want to perform the startup and shutdown of
 # the video stream during any given day. A value of zero would mean exactly on
```

and CrowCam.sh - to call the API instead of the helper function GetSunriseSunsetTimeFromGoogle

```
$ diff -u CrowCam\ -\ Backup.sh CrowCam.sh
--- "CrowCam - Backup.sh"       2024-07-17 15:05:43.890000000 +0800
+++ CrowCam.sh  2025-01-29 16:05:20.274000000 +0800
@@ -1048,21 +1048,28 @@
   if [ ! -f "$crowcamSunrise" ] || [ ! -f "$crowcamSunset" ] || [ "$( find "$crowcamSunrise" -mmin +$ageLimit )" != "" ] || [ "$( find "$crowcamSunset" -mmin +$ageLimit )" != "" ]
   then
     # Get more accurate sunrise/sunset results from Google if available.
-    LogMessage "dbg" "Retrieving sunrise/sunset times from Google"
-    googleSunriseString=$(GetSunriseSunsetTimeFromGoogle "Sunrise")
-    googleSunsetString=$(GetSunriseSunsetTimeFromGoogle "Sunset")
+#    LogMessage "dbg" "Retrieving sunrise/sunset times from Google"
+#    googleSunriseString=$(GetSunriseSunsetTimeFromGoogle "Sunrise")
+#    googleSunsetString=$(GetSunriseSunsetTimeFromGoogle "Sunset")
+
+       LogMessage "dbg" "Retrieving sunrise/sunset from api.sunrisesunset.io"
+       jsonresults=$(wget -qO- "https://api.sunrisesunset.io/json?lat=$lat&lng=$lng")
+       googleSunriseString=$(echo $jsonresults | jq -r .results.sunrise)
+    googleSunsetString=$(echo $jsonresults | jq -r .results.sunset)

     # Make sure the Google responses were non-null and non-empty.
     if [ -z "$googleSunriseString" ] || [ -z "$googleSunsetString" ]
     then
       # Failure condition, did not retrieve values.
-      LogMessage "err" "Problem obtaining sunrise/sunset from Google. Falling back to previously saved values: $sunrise/$sunset"
+      #LogMessage "err" "Problem obtaining sunrise/sunset from Google. Falling back to previously saved values: $sunrise/$sunset"
+      LogMessage "err" "Problem obtaining sunrise/sunset from api.sunrisesunset.io. Falling back to previously saved values: $sunrise/$sunset"
     else
       # Success condition. If the Google responses were non-empty, use them
       # in place of the fallback values and write them to the fallback files.
       sunrise=$googleSunriseString
       sunset=$googleSunsetString
-      LogMessage "info" "Retrieved sunrise/sunset times from Google: $sunrise/$sunset"
+     # LogMessage "info" "Retrieved sunrise/sunset times from Google: $sunrise/$sunset"
+      LogMessage "info" "Retrieved sunrise/sunset times from api.sunrisesunset.io: $sunrise/$sunset"

       # If the Google responses were not empty, then write them into our fallback
       # files. Use "-n" to ensure they do not write a trailing newline character.
@@ -1079,7 +1086,8 @@
 # sunrise or sunset times in these variables.
 if [ -z "$sunrise" ] || [ -z "$sunset" ]
 then
-  LogMessage "err" "Problem obtaining sunrise/sunset values. Unable to obtain values from Google, and also unable to use fallback values"
+  #LogMessage "err" "Problem obtaining sunrise/sunset values. Unable to obtain values from Google, and also unable to use fallback values"
+  LogMessage "err" "Problem obtaining sunrise/sunset from api.sunrisesunset.io. Falling back to previously saved values: $sunrise/$sunset"
   exit 1
 fi
```

As you can see, the core of the change is
```
+       jsonresults=$(wget -qO- "https://api.sunrisesunset.io/json?lat=$lat&lng=$lng")
+       googleSunriseString=$(echo $jsonresults | jq -r .results.sunrise)
+       googleSunsetString=$(echo $jsonresults | jq -r .results.sunset)
```

I hope this helps.

Regards
Damian

tfabris, 2025-01-29:
Thank you so much for the report and for the code change alternative. 

This is a duplicate of issue #95 which was already fixed last week with checkin https://github.com/tfabris/CrowCam/commit/4550e6671fe8b2d155e6ea13444d85e49ec1c83b

I’d already considered and rejected sunrisesunset.io for my own reasons, though I’ll have a look at how you’re implementing it and see if it’s got advantages. I’ll look at it tomorrow some time. 

Thank you! And I’m glad to see folks using the code!

quog, 2025-01-29:
> last week with checkin 4550e66

Hi Tony,

I hadn't seen your fix or I would have grabbed it instead.  However, in my
view, using an api like sunrisesunset.io is much cleaner than screen
scraping either Google or timeanddate.com - both are subject to them
changing their screen layout.

However, it's your code - I am just happy to contribute but if you want to
use it or not it is up to you.

Regards
Damian

Damian Douglas-Meyer
0414 419 624
***@***.***

tfabris, 2025-02-12:
@quog - Thanks again for your suggestion. I implemented a variation on it, and it's checked in now, with this commit: [0a8d3bf](https://github.com/tfabris/CrowCam/commit/0a8d3bf3106fefb39a9bc52b6991144d100c3af1)

The code changes needed to be more complicated than your suggestion, for several different reasons, but feel free to look at the code diffs and see what I did.

Thanks so much!

closed 2025-02-12 by tfabris

tfabris, 2025-02-27:
@quog - You might want to pick up the most recent code from the master branch, I just fixed an intermittent bug in my JSON response parsing (a secondary side effect of my stubborn refusal to use JQ). See the bugfixes I made for #92 and #93. Unrelated to the sunrise/sunset stuff but which will continue cropping up intermittently if you're not using the latest code.